OpenMOSS/VLABench #1: Schedule for release

Fisher-Wang, 2024-12-31:
Hi, thank you for your amazing work which contribute a comprehensive benchmark for manipulation tasks!

I wanted to kindly ask is there any schedule for releasing the rest of the code and data, like task definition, demonstration trajectories, ...? I look forward to the full release ;)

Shiduo-zh, 2025-01-06:
Hi! Thanks for your support! The full scripts will be released in late January or early February, after thorough testing. The formal release will include the complete task loading way, an automated data framework, convenient evaluation tools, teleoperation, and a standard and useful dataset for finetuning  (covering both trajectory and reasoning data).

I will also add a benchmark setting for imitation learning, along with a full leaderboard of policies and VLMs across different settings. At the same time, I will conduct more real-world experiments to ensure that the simulation data has the potential for data augmentation as much as possible. I will conduct reliable experiments for the official comprehensive leaderboard as soon as possible, and expect to have it ready by the end of February at the latest.

closed 2025-02-12 by Shiduo-zh

jasper0314-huang, 2025-02-15:
Hi @Shiduo-zh ,

Thanks for your great work! I saw that the planned release was expected in early February. I was wondering if the release schedule remains the same or if there are any updates on the timeline. Looking forward to the full release!

Thanks!

Shiduo-zh, 2025-02-17:
> Hi [@Shiduo-zh](https://github.com/Shiduo-zh) ,
> 
> Thanks for your great work! I saw that the planned release was expected in early February. I was wondering if the release schedule remains the same or if there are any updates on the timeline. Looking forward to the full release!
> 
> Thanks!

Hi Jasper! I’m currently working on the reconstruction and testing of the new code. Here’s an update: I have committed the basic evaluation script for action policies and VLMs, along with scripts for auto trajectory generation for most tasks. However, the testing is not yet robust enough. I will update the tutorial and README after further testing, which I plan to complete in about a week. The overall schedule may be delayed by one or two weeks. Thanks for your support!