OpenMOSS/VLABench #6: Convert to rlds/relobot

4078730, 2025-03-01:
Thank you for releasing such an excellent project!

I understand that this is still a pre-release version, but is there any publicly available method for converting the collected trajectories into RLDS or LeRobot dataset formats?

I'd like to try fine-tuning VLA with the collected data.

Thank you

Shiduo-zh, 2025-03-02:
Hi, thank you for your support! I will update a tutorial on how to convert our data to Lerobot and rlds data format about one recently(about March 10th). You can refer to [rlds_dataset_builder](https://github.com/kpertsch/rlds_dataset_builder) for rlds format generation and the libero scripts in [pi0](https://github.com/Physical-Intelligence/openpi/blob/main/examples/libero/convert_libero_data_to_lerobot.py)

4078730, 2025-03-03:
Thank you! I'm looking forward to the tutorial release around March 10th!
In the meantime, I'll refer to the rlds_dataset_builder and the libero scripts in pi0 as you suggested.

closed 2025-03-03 by 4078730