OpenMOSS/VLABench #11: mujoco.FatalError: gladLoadGL error

XilunZhangRobo, 2025-04-04:
Hi, 
Thanks for releasing your code! 
While I am generating the dataset with dataset_generation.sh, I encountered the following error: 
```
Traceback (most recent call last):                                                                                                                                                                  
  File "/mnt/new-ssd/extra-space/xilun/VLABench/scripts/trajectory_generation.py", line 124, in <module>                                                                                            
    generate_trajectory(args, i, logger)                                                                                                                                                            
  File "/mnt/new-ssd/extra-space/xilun/VLABench/scripts/trajectory_generation.py", line 68, in generate_trajectory                                                                                  
    obs, waypoint, stage_success, task_success = skill(env)                                                                                                                                         
  File "/mnt/new-ssd/extra-space/xilun/VLABench/VLABench/utils/skill_lib.py", line 139, in pick                                                                                                     
    key_pos, prepare_key_pos, key_quat = find_keypoint_and_prepare_grasp(env, target_entity, prior_eulers, specific_keypoint_id=specific_keypoint, move_vector=prepare_quat)                        
  File "/mnt/new-ssd/extra-space/xilun/VLABench/VLABench/utils/utils.py", line 126, in find_keypoint_and_prepare_grasp                                                                              
    env_pcd = env.get_observation()["masked_point_cloud"]                                                                                                                                           
  File "/mnt/new-ssd/extra-space/xilun/VLABench/VLABench/envs/dm_env.py", line 120, in get_observation                                                                                              
    multi_view_rgb.append(self.render(camera_id=cam, **self.render_options))                                                                                                                        
  File "/mnt/new-ssd/extra-space/xilun/VLABench/VLABench/envs/dm_env.py", line 32, in render                                                                                                        
    return self.physics.render(**kwargs)                                                                                                                                                            
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/mujoco/engine.py", line 223, in render                                                                         
    camera = Camera(                                                                                                                                                                                
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/mujoco/engine.py", line 711, in __init__                                                                       
    if self._physics.contexts.mujoco is not None:                                                                                                                                                   
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/mujoco/engine.py", line 533, in contexts                                                                       
    self._make_rendering_contexts()                                                                                                                                                                 
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/mujoco/engine.py", line 519, in _make_rendering_contexts                                                       
    mujoco_context = wrapper.MjrContext(self.model, render_context)                                                                                                                                 
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/mujoco/wrapper/core.py", line 608, in __init__                                                                 
    ptr = ctx.call(mujoco.MjrContext, model.ptr, font_scale)                                                                                                                                        
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/dm_control/_render/executor/render_executor.py", line 138, in call                                                        
    return func(*args, **kwargs)                                                                                                                                                                    
mujoco.FatalError: gladLoadGL error
```

Is it because I do not have a valid renderer since I am running it on the server through ssh? 

Thanks

Shiduo-zh, 2025-04-05:
One possible solution is to set the environment variable `MUJOCO_GL=egl`. 

If this doesn't work, please make sure that the system has the necessary EGL-related dependencies installed:
```sh
sudo apt-get install mesa-utils  
sudo apt-get install libglu1-mesa  
sudo apt-get install libgl1-mesa-dri  
sudo apt-get install libgl1-mesa-glx
```

XilunZhangRobo, 2025-04-05:
Ok, it works now. Thanks

closed 2025-04-05 by XilunZhangRobo