OpenMOSS/VLABench #13: Could not load DatasetBuilder

XilunZhangRobo, 2025-04-07:
Hi, 
After I generated the hdf5 for several tasks such as select_toy and select_fruit, it seems that 
`python scripts/convert_to_rlds.py --task [list] --save_dir /your/path/to/dataset
`
cannot take multiple tasks at the same time. So I run the task with "select_toy". 
Then I cd into the select_toy data folder to run the `tfds build`. 
It raises an error: 
```
ValueError: Could not load DatasetBuilder from: select_toy. Make sure the module only contains a single `DatasetBuilder`.
If no dataset is detected, make sure that all abstractmethods are implemented.
Detected builders: []
```
Do you have any insights on this issue? 

Thanks,
Xilun

Shiduo-zh, 2025-04-08:
Hello! The script `scripts/convert_to_rlds.py` will create a Python file named after the task_name in the corresponding HDF5 directory. This file includes the adapted tfds_builder class. You can first check if the `select_toy.py` file has been correctly generated in the `{dataset_root}/select_toy` directory.

XilunZhangRobo, 2025-04-11:
I have the select_toy.py generated in the directory. Here is the complete error message: 
`(vlabench) xilun@crazycrab:~/extra-space/VLABench/dataset/select_toy$ tfds build
2025-04-11 12:31:56.339968: I tensorflow/core/util/port.cc:153] oneDNN custom operations are on. You may see slightly different numerical results due to floating-point round-off errors from different computation orders. To turn them off, set the environment variable `TF_ENABLE_ONEDNN_OPTS=0`.
2025-04-11 12:31:56.347830: E external/local_xla/xla/stream_executor/cuda/cuda_fft.cc:467] Unable to register cuFFT factory: Attempting to register factory for plugin cuFFT when one has already been registered
WARNING: All log messages before absl::InitializeLog() is called are written to STDERR
E0000 00:00:1744399916.356536  933169 cuda_dnn.cc:8579] Unable to register cuDNN factory: Attempting to register factory for plugin cuDNN when one has already been registered
E0000 00:00:1744399916.359097  933169 cuda_blas.cc:1407] Unable to register cuBLAS factory: Attempting to register factory for plugin cuBLAS when one has already been registered
W0000 00:00:1744399916.366029  933169 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1744399916.366039  933169 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1744399916.366040  933169 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
W0000 00:00:1744399916.366042  933169 computation_placer.cc:177] computation placer already registered. Please check linkage and avoid linking the same target more than once.
2025-04-11 12:31:56.368125: I tensorflow/core/platform/cpu_feature_guard.cc:210] This TensorFlow binary is optimized to use available CPU instructions in performance-critical operations.
To enable the following instructions: AVX2 AVX_VNNI FMA, in other operations, rebuild TensorFlow with the appropriate compiler flags.
INFO[build.py]: Loading dataset  from path: /mnt/new-ssd/extra-space/xilun/VLABench/dataset/select_toy/select_toy.py
Traceback (most recent call last):
  File "/home/xilun/miniconda3/envs/vlabench/bin/tfds", line 8, in <module>
    sys.exit(launch_cli())
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/scripts/cli/main.py", line 113, in launch_cli
    app.run(main, flags_parser=_parse_flags)
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/absl/app.py", line 316, in run
    _run_main(main, args)
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/absl/app.py", line 261, in _run_main
    sys.exit(main(argv))
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/scripts/cli/main.py", line 108, in main
    args.subparser_fn(args)
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/scripts/cli/build.py", line 111, in _build_datasets
    builders_cls_and_kwargs = [
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/scripts/cli/build.py", line 112, in <listcomp>
    _get_builder_cls_and_kwargs(dataset, has_imports=bool(args.imports))
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/scripts/cli/build.py", line 222, in _get_builder_cls_and_kwargs
    builder_cls = tfds.core.community.builder_cls_from_module(path.stem)
  File "/home/xilun/miniconda3/envs/vlabench/lib/python3.10/site-packages/tensorflow_datasets/core/community/load.py", line 55, in builder_cls_from_module
    raise ValueError(
ValueError: Could not load DatasetBuilder from: select_toy. Make sure the module only contains a single `DatasetBuilder`.
If no dataset is detected, make sure that all abstractmethods are implemented.
Detected builders: []`

Shiduo-zh, 2025-04-14:
Hello! This situation seems a bit strange. Could you please attach the `select_toy.py` file to this issue? I will try to reproduce the problem.

XilunZhangRobo, 2025-04-16:
I have pasted the generated code here. 
Thanks! 

`from typing import Tuple, Any, Dict, Union, Callable, Iterable, Iterator
import numpy as np
import tensorflow as tf
import tensorflow_datasets as tfds

import itertools
from multiprocessing import Pool
from functools import partial
import glob
import h5py
import tensorflow as tf
import tensorflow_hub as hub
from tensorflow_datasets.core import download
from tensorflow_datasets.core import split_builder as split_builder_lib
from tensorflow_datasets.core import naming
from tensorflow_datasets.core import splits as splits_lib
from tensorflow_datasets.core import utils
from tensorflow_datasets.core import writer as writer_lib
from tensorflow_datasets.core import example_serializer
from tensorflow_datasets.core import dataset_builder
from tensorflow_datasets.core import file_adapters

Key = Union[str, int]
# The nested example dict passed to `features.encode_example`
Example = Dict[str, Any]
KeyExample = Tuple[Key, Example]

class SelectToy(tfds.core.GeneratorBasedBuilder):
    """DatasetBuilder for example dataset."""
    VERSION = tfds.core.Version('1.0.0')
    RELEASE_NOTES = {
      '1.0.0': 'Initial release.',
    }
    N_WORKERS = 10             # number of parallel workers for data conversion
    MAX_PATHS_IN_MEMORY = 500

    def __init__(self, *args, **kwargs):
        self.path = "*.hdf5"
        super().__init__(*args, **kwargs)
    
    def _split_paths(self):
        """Define filepaths for data splits."""
        return {
            'train': glob.glob('*.hdf5'),
        }
    
    def _split_generators(self, dl_manager: tfds.download.DownloadManager):
        """Define data splits."""
        return {
            'train': self._generate_examples(path=self.path),
        }

    def _generate_examples(self, path) -> Iterator[Tuple[str, Any]]:
        """Generator of examples for each split."""

        def _parse_example(episode_path):
            # load raw data --> this should change for your dataset
            # data = np.load(episode_path, allow_pickle=True)     # this is a list of dicts in our case
            with h5py.File(episode_path, 'a') as f:
                data = f["data"]
                timestamps = data.keys()
                for i, ts in enumerate(timestamps):
                    new_data = data[ts]
                    trajectory = np.asarray(new_data['trajectory'])
                    images = np.asarray(new_data["observation"]["rgb"])
                    # qpos = np.asarray(new_data["observation"]["qpos"])
                    # ee_state = np.asarray(new_data["observation"]["ee_state"])
[... 275 lines cut ...]
        split_info = splits_lib.SplitInfo(
            name=split_name,
            shard_lengths=shard_lengths,
            num_bytes=total_size,
            filename_template=filename_template,
        )
        return _SplitInfoFuture(lambda: split_info)

def dictlist2listdict(DL):
    " Converts a dict of lists to a list of dicts "
    return [dict(zip(DL, t)) for t in zip(*DL.values())]

def chunks(l, n):
    """Yield n number of sequential chunks from l."""
    d, r = divmod(len(l), n)
    for i in range(n):
        si = (d + 1) * (i if i < r else r) + d * (0 if i < r else i - r)
        yield l[si:si + (d + 1 if i < r else d)]

def chunk_max(l, n, max_chunk_sum):
    out = []
    for _ in range(int(np.ceil(len(l) / max_chunk_sum))):
        out.append(list(chunks(l[:max_chunk_sum], n)))
        l = l[max_chunk_sum:]
    return out`

closed 2025-06-05 by Shiduo-zh