OpenMOSS/VLABench #14: Could not select Xarm as robot type

wzrrr1016, 2025-04-07:
Hi. Thank you for your excellent work!
I ran the the first tutorial 1.load_task.ipynb and I selected xarm as the robot type.
The lines are as follow:
`task = "select_fruit"
robot = "xarm"
env = load_env(task, robot=robot, time_limit=1000)
env.reset()

image = env.render(camera_id=2, width=640, height=640)
Image.fromarray(image)`

I got error below.
It seems like the code for Xarm is not ready yet.

Look forward to your reply!

`---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[29], [line 2](vscode-notebook-cell:?execution_count=29&line=2)
      [1](vscode-notebook-cell:?execution_count=29&line=1) env = load_env(task, robot=robot, time_limit=1000)
----> [2](vscode-notebook-cell:?execution_count=29&line=2) env.reset()
      [4](vscode-notebook-cell:?execution_count=29&line=4) image = env.render(camera_id=2, width=640, height=640)
      [5](vscode-notebook-cell:?execution_count=29&line=5) Image.fromarray(image)

File /workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:21, in LM4ManipDMEnv.reset(self)
     [19](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:19) def reset(self):
     [20](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:20)     self.timestep = 0
---> [21](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:21)     timestep = super().reset()
     [22](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:22)     self.cancel_gravity_and_improve_fluid()
     [23](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:23)     for i in range(self.reset_wait_step):
     [24](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:24)         # print(f"reset {i}")

File /usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:367, in Environment.reset(self)
    [365](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:365) while True:
    [366](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:366)   try:
--> [367](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:367)     return self._reset_attempt()
    [368](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:368)   except EpisodeInitializationError as e:
    [369](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:369)     failed_attempts += 1

File /workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:186, in LM4ManipDMEnv._reset_attempt(self)
    [184](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:184) self.task.reset_distractors(n_distractor=self.n_distractor)
    [185](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:185) self._hooks.refresh_entity_hooks()
--> [186](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/envs/dm_env.py:186) return super()._reset_attempt()

File /usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:386, in Environment._reset_attempt(self)
    [384](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:384)   self._random_state.set_state(self._fixed_random_state)
    [385](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:385) with self._physics.reset_context():
--> [386](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:386)   self._hooks.initialize_episode(self._physics_proxy, self._random_state)
    [387](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:387) self._observation_updater.reset(self._physics_proxy, self._random_state)
    [388](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:388) self._reset_next_step = False

File /usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:128, in _EnvironmentHooks.initialize_episode(self, physics, random_state)
    [126](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:126) self._task.initialize_episode(physics, random_state)
    [127](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:127) for entity_hook in self._initialize_episode.entity_hooks:
--> [128](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:128)   entity_hook(physics, random_state)
    [129](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:129) for extra_hook in self._initialize_episode.extra_hooks:
    [130](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/composer/environment.py:130)   extra_hook(physics, random_state)

File /workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:38, in XArm.initialize_episode(self, physics, random_state)
     [36](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:36) super().initialize_episode(physics, random_state)
     [37](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:37) for i, qpos in enumerate(self.default_qpos):
---> [38](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:38)     if i < 7: physics.bind(self.mjcf_model.find("joint", f"act{i + 1}")).qpos = qpos
     [39](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:39)     else: physics.bind(self.mjcf_model.find("joint", f"gripper")).qpos = qpos
     [40](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/workspace/robotics/home_wzr/benchmark/VLABench/VLABench/robots/single_arm/xarm.py:40) physics.data.ctrl = self.default_qpos

[... cut ...]
    [620](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/mjcf/physics.py:620) 
    [621](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/mjcf/physics.py:621) # Checking for iterability in this way is cheaper than using `isinstance`.
    [622](https://vscode-remote+ssh-002dremote-002b172-002e21-002e129-002e49.vscode-resource.vscode-cdn.net/usr/local/lib/python3.10/dist-packages/dm_control/mjcf/physics.py:622) try:

ValueError: mjcf_elements is None.`

[image]

Shiduo-zh, 2025-04-08:
Hello! Thank you for your interest in our work. Due to ongoing iterations of the codebase, additional interface modifications and testing are still required for other types of robotic arms, including Xarm and WidowX. This process is expected to be completed within April, as it involves updates and adaptations to the general interface. I’ll reply to this issue again once the update is released.

wzrrr1016, 2025-04-09:
Thanks. Look forward to your update!

closed 2025-04-14 by Shiduo-zh