OpenMOSS/VLABench #15: Reproduce the finetuning datasets in paper

Facebear-ljx, 2025-04-08:
Hi, VLAbench is one impressive work! I want to evaluate my model on VLAbench and conduct fair comparisons with the results of RDT, OpenVLA and Octo in your paper (table 2). So, I am wondering if you can provide the data generation code to help me reproduce the finetuning datasets used in your paper?

Facebear-ljx, 2025-04-08:
Is the huggingface dataset used for finetuning? I noticed that the huggingface dataset contains 500 episodes for 10 tasks, while the paper reports that 100 tasks for 16 tasks are used.

Shiduo-zh, 2025-04-08:
Hello! Thank you for your interest in and recognition of our work!

Yes, the dataset in Hugging Face is intended for fine-tuning and is available for everyone to use. You can use it directly! The 500 episodes are part of the updated experimental setting, which will be reflected in Paper v2 (likely in May). These 500 data points slightly differ from the v1 experimental data, as we strictly controlled the consistency of the background to ensure the data distribution is closely aligned. This was done to ensure multi-dimensional generalization testing during evaluation (refer to tracks 1-6).

Although the experimental results have been updated, the conclusions remain largely consistent with those in the v1 paper. In this update, we also added pi0, which performs significantly better than the previous models.

Facebear-ljx, 2025-04-09:
Cool! Looking forward to the updated version :)

closed 2025-04-09 by Facebear-ljx