OpenMOSS/VLABench #16: Typo in readme

LukeLIN-web, 2025-04-09:
https://github.com/OpenMOSS/VLABench/blob/4f5b5ee691701fb077eb558b8968e4968cad309d/README.md?plain=1#L30

should be `python scripts/download_assets.py `

Shiduo-zh, 2025-04-09:
Thank you for the reminder! I’ll fix these spelling errors in the next update.

LukeLIN-web, 2025-04-09:
> Thank you for the reminder! I’ll fix these spelling errors in the next update.

In [--loar_ckpt xx](https://github.com/OpenMOSS/VLABench/tree/main?tab=readme-ov-file#vlapolicy-evaluation)  
`--loar_ckpt xx` is typo

XilunZhangRobo, 2025-04-11:
Hi, I did not find the uploaded OpenVLA checkpoints on hugging face. Are you planning to upload the fine-tuned checkpoint on hugging face?

XilunZhangRobo, 2025-04-11:
One minor typo I found, It seems like you do not have `--with-cot` argument in `scripts/evaluate_vlm.py`.

closed 2025-08-06 by Shiduo-zh