OpenMOSS/VLABench #17: What is the usage of `sh data_generation.sh `

LukeLIN-web, 2025-04-10:
1. I am confused,  This data is necessary for inference? Or use for training? 
2. Is "VLABench/vlabench_primitive_ft_dataset" is used for training ?

Shiduo-zh, 2025-04-10:
Hi there!
Q1: This script provides the way we run parallel data collection. All successfully collected data can be used for model training. For standardized evaluation, we provide fair episodes for comparing different policies under `VLABench/configs/evaluation/tracks`.

Q2: Yes, the dataset on Hugging Face is the one we provide for fine-tuning and unified comparison.

LukeLIN-web, 2025-04-11:
When I try to load it  `from datasets import load_dataset

# Login using e.g. `huggingface-cli login` to access this dataset
ds = load_dataset("VLABench/vlabench_primitive_ft_dataset")`
I meet 
```
vlabench_primitive.tar.gz.76: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.29G/1.29G [00:12<00:00, 104MB/s]
Downloading data: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 77/77 [2:09:50<00:00, 101.18s/files]s]
Generating train split: 101 examples [01:24,  1.19 examples/s]
Traceback (most recent call last):
  File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/builder.py", line 1608, in _prepare_split_single
    for key, record in generator:
  File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/packaged_modules/webdataset/webdataset.py", line 122, in _generate_examples
    for example_idx, example in enumerate(self._get_pipeline_from_tar(tar_path, tar_iterator)):
  File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/packaged_modules/webdataset/webdataset.py", line 44, in _get_pipeline_from_tar
    current_example[field_name.lower()] = f.read()
  File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/tarfile.py", line 691, in read
    raise ReadError("unexpected end of data")
tarfile.ReadError: unexpected end of data
```

closed 2025-04-16 by LukeLIN-web

EmbodiedAI-Group, 2025-05-20:
> When I try to load it by `load_dataset` I meet error
> 
> ```
> from datasets import load_dataset
> # Login using e.g. `huggingface-cli login` to access this dataset
> ds = load_dataset("VLABench/vlabench_primitive_ft_dataset")
> 
> 
> vlabench_primitive.tar.gz.76: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1.29G/1.29G [00:12<00:00, 104MB/s]
> Downloading data: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 77/77 [2:09:50<00:00, 101.18s/files]s]
> Generating train split: 101 examples [01:24,  1.19 examples/s]
> Traceback (most recent call last):
>   File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/builder.py", line 1608, in _prepare_split_single
>     for key, record in generator:
>   File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/packaged_modules/webdataset/webdataset.py", line 122, in _generate_examples
>     for example_idx, example in enumerate(self._get_pipeline_from_tar(tar_path, tar_iterator)):
>   File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/site-packages/datasets/packaged_modules/webdataset/webdataset.py", line 44, in _get_pipeline_from_tar
>     current_example[field_name.lower()] = f.read()
>   File "/home/user1/anaconda3/envs/vlabench/lib/python3.10/tarfile.py", line 691, in read
>     raise ReadError("unexpected end of data")
> tarfile.ReadError: unexpected end of data
> ```
> 
> For someone may want to know.
> 
> I download https://huggingface.co/datasets/VLABench/vlabench_primitive_ft_dataset
> 
> ```
> git clone  https://huggingface.co/datasets/VLABench/vlabench_primitive_ft_dataset  
> git lfs pull
> 
> 1.5T    ./.git
> 2.3T    .
> ```
> 
> it is huge.

I have met the same problem!  It's too hard to download the dataset. Any other convenient way to download the dataset?