OpenMOSS/VLABench #19: Abs EEF Action Space or Abs Joint Action Space

Facebear-ljx, 2025-04-11:
Hi, I trained my model on the huggingface dataset using `trajectory` as the key for action prediction, which seems to be absolute eef pos of the robot. So in evaluation, I set the control mode as `eef`. I checked the output of my model and `observation['ee_state`] with the first 3 dimension minus [0, -0.4, 0.78] and quat to euler transformation according to your data process code. The output of my model looks good, which looks like a smooth future eef trajectory starting from the current ee_state:

[image]

However, the resulting rollouts are very suboptimal:

https://github.com/user-attachments/assets/f3183312-5473-490b-8c6f-f9c694c0da44

I suspect the reason behind this is the potential transformation error when transforming the abs eef trajectory to joint pos. So, I am wondering whether you use abs eef trajectory or joint pos to control the robot? It would be very appreciated if you could help me resolve this issue.

Shiduo-zh, 2025-04-11:
Hi, the control method used in VLABench is based on the inverse kinematics solver from`dm_control`, which maps the end-effector state (ee_state) to joint positions via IK (it also supports direct joint position control). 

One possible reason for the issue you’re encountering is that the IK interface accepts positions in the world coordinate system. Therefore, you can try using `ee_state + delta_action` directly without subtracting the Franka base coordinates [0, -0.4, 0.78].

Facebear-ljx, 2025-04-12:
Thanks for your answer! The rollout runs normal now:)

closed 2025-04-12 by Facebear-ljx

nansun5410, 2026-01-12:
> Thanks for your answer! The rollout runs normal now:)

Have you solve this problem?  I train the abs ee pose from the lerobot version dataset on Qwen3VL with fast tokenizer, my results are similar to yours, but i have add the offset on my output before sending to the simulator. How do you make your rollouts normal? I'll be grateful if you can reply, thanks.