OpenMOSS/VLABench #22: Six Benchmark results

lzl2040, 2025-04-13:
Thanks for your work! I do not find the relevant results on the six benchmark in the paper. What is the difference between the six benchmark and Table 2? Besides, how to use the current code to get the results from Table 2?

Shiduo-zh, 2025-04-14:
Hi! The six dimension benchmark is designed for VLM evaluation, and its evaluation script is `scripts/evaluate_vlm.py`. Table 2 presents the evaluation results of the VLA policies, organized by tasks. We provide multi-dimensional generalization capabilities for the VLA evaluation, referencing the six tracks in the README. To obtain similar results as in Table 2, please refer to the `VLA/policy evaluation` section in the README to evaluate the fine-tuned VLA.

Zenobiaql, 2025-04-14:
Thanks for you reply! Besides, the results of Table 2 is obtained by fine-tuning the model on [this dataset](https://huggingface.co/datasets/VLABench/vlabench_primitive_ft_dataset)?

Shiduo-zh, 2025-04-15:
The results of Table 2 is the first version of the paper, with each task having 100 episodes for fine-tuning. The dataset provided in the huggingface provides 500 episodes for each tasks, and we updated the evaluation result of pi0 in the v2 paper(update later). The performance of pi0 is significantly better than other models, and the results using our data on models like OpenVLA/RDT show performances consistent with v1.

lzl2040, 2025-04-16:
Thanks! If I want to obtain the pi0 results in Table 2(a), do I need to fine-tune the model on all tasks of [this dataset](https://huggingface.co/datasets/VLABench/vlabench_primitive_ft_dataset) first? Then, which track should I choose for evaluation in order to get the results shown in Table 2(a)? And for the results of Table 2(b), (c) and (d)?

closed 2025-11-11 by Shiduo-zh