OpenMOSS/VLABench #24: pi0 inference in VLABench's mujoco environment

Fuego910718, 2025-04-14:
Thanks for your work! Can you come up with a scripts about pi0's INFERENCE on the VLABench dataset? Also, in the file `eval_pi0.sh`, where is the part of args.episode_config_path that needs to be replaced? Is it a file like **`/VLABench/VLABench/configs/evaluation/tracks/track_1_in_distribution.json`**? Also I don't know where the task's INSTRUCTION is given? Looking forward to your reply!

Shiduo-zh, 2025-04-15:
Hi! We provide a reference evaluation pipeline following pi0 in https://github.com/Shiduo-zh/openpi/blob/main/examples/vlabench/README.md, you should run a server policy first and run the `eval_pi0.sh` in another terminal.

`args.episode_config_path` refers to the path of different json files in VLABench/configs/evaluation/tracks. The instructions will be loaded from these evaluation episode configs.

Fuego910718, 2025-04-18:
Thanks for you reply! Today I am evaluating according to the reference you gave me and when executing `eval_pi0.sh`, I keep getting the error message ‘**`ModuleNotFoundError: No module named “plotly”`**’.  But I have activated the .venv environment, and I have `plotly==6.0.0` in the pip list, but I still get the error. Is there any solution for this? I don't see any information about `plotly` in `requirement.txt`. Looking forward to your reply!

JYounngS, 2025-06-10:
I encountered the same problem. Has anyone successfully solved it?

Rubiksman78, 2025-06-19:
Installing the dependencies of VLABench with uv used in the venv created `eval_pi0.sh` seems to have solved the issue for me.

Shiduo-zh, 2025-11-11:
I would like to recommend using uv to run openpi policy and using conda environment to run vlabench, just refers to  https://github.com/Shiduo-zh/openpi/blob/main/multi_run_vlabench.sh and https://github.com/Shiduo-zh/openpi/blob/main/serve_policy.sh.

closed 2025-11-11 by Shiduo-zh