OpenMOSS/VLABench #26: Experimental reproduction of Table 2

abliao, 2025-04-18:
How can I reproduce the results of Table 2 in the paper?

Shiduo-zh, 2025-04-21:
Hello, in the recent update, we have split the 4 sub-tables in Table 2 into multiple evaluation tracks. Please refer to the README for an introduction to the different tracks. To reproduce a similar experiment results, please refer to [issue 15](https://github.com/OpenMOSS/VLABench/issues/15).

abliao, 2025-04-23:
Thanks for your answer. I would also like to know what the new experimental results are, because [issue 15](https://github.com/OpenMOSS/VLABench/issues/15) mentioned that the results are different from the v1 paper.

Shiduo-zh, 2025-11-11:
Please refers to the model's results listed in https://huggingface.co/VLABench/models.