OpenMOSS/VLABench #28: convert_to_lerobot requirements

PKU-ShanYachun, 2025-04-19:
The Lerobot version used in scripts/convert_to_lerobot can not be downloaded from git now. The released version of lerobot is incompatible with the scripts requirements.

Shiduo-zh, 2025-04-21:
Hello, this version of the `convert_to_lerobot` script is based on the method used by pi0 to generate the Libero Lerobot dataset. We will develop a new version of the Lerobot script as soon as possible. Thank you for your feedback!

chen-xin-94, 2025-06-05:
Same issue here. Any update? :-)

mxjbj, 2025-07-11:
手动 clone 安装 lerobot

git clone https://github.com/huggingface/lerobot
cd lerobot
git checkout 6674e368249472c91382eb54bb8501c94c7f0c56

# 2. 打开 pyproject.toml，找到：
#    pyav = ">=12.0.5"
#    改成：
#    av = ">=12.0.5"

nano pyproject.toml   # 或 vim、gedit 等文本编辑器

pip install av
 
pip install -e .