OpenMOSS/VLABench #31: WARNING:absl:Failed to converge after 99 steps: err_norm=0.0868166

Baymax-go, 2025-04-25:
I keep getting this warning when running randompolicy. Can I ignore it? Or can you explain what causes it?

Shiduo-zh, 2025-04-30:
This issue is caused by the given end-effector position exceeding the maximum limits, which leads to the inverse kinematics solver encountering the error mentioned above. If this issue occurs during policy inference, it most likely indicates that the policy output is not as expected.

kingjulio8238, 2025-05-05:
Any suggestions on how to fix this? I am using the base openvla-7B model.

RonPlusSign, 2025-05-12:
I'm facing the same issue. I'm trying to replicate the results of OpenVLA from the paper, but I have the same warning in almost all of the given tasks, with some tasks that even give some errors (e.g. `insert_flower` gives me an "invalid physics state" error).
I also see that some task names are not registered (I see the task names as mentioned in [#27](https://github.com/OpenMOSS/VLABench/issues/27)), e.g. there's no `select_book_common_sense`. How can I test these tasks to complete the first table of results?
How many episodes are used in the paper for each task? I cannot exactly replicate the results without knowing this

Rubiksman78, 2025-06-19:
Same issue with both versions of pi0 from `VLABench/pi0-base-primitive` and `VLABench/pi0-fast-primitive` on track 1.

Shiduo-zh, 2025-08-06:
I want to clarify that this is not a bug, but rather a failure of the inverse kinematics solution caused by the robotic arm exceeding its working limits due to unreasonable model outputs.
In summary, this issue occurs because of the model’s improper output. Usually, a better policy will result in fewer occurrences of this warning.

closed 2025-08-06 by Shiduo-zh