OpenMOSS/VLABench #36: Error: mesh volume is too small

PKU-ShanYachun, 2025-05-13:
When I use VLABench as a client to inference on pi0 server,  I get the error below. I download assets as readme. Why the simulate environment pops this error?

Error: mesh volume is too small: pan_seen/dishes/dish_without_plate_2_0 . Try setting inertia to shell
Element name 'pan_seen/dishes/dish_without_plate_2_0', id 188, line 415
<mesh name="pan_seen/dishes/dish_without_plate_2_0" class="pan_seen/dishes/" file="dish_without_plate_2_0-a8cbb37bcf6f46466a4b6585030ad7eb2639e0cd.obj"/>

PKU-ShanYachun, 2025-05-13:
The problem comes out when I evaluate the add_condiment task solely
.

Shiduo-zh, 2025-06-05:
Sorry for the late reply — things have been quite busy lately. I believe this issue is related to the way the mesh is processed. I’m planning to simplify some of the more complex meshes soon, as overly detailed meshes can negatively affect the stability of the simulation. Stay tuned!