OpenMOSS/VLABench #64: The version of lerobot

xiaokang-99, 2025-08-31:
大佬您好，我在进行pip install -r requirements.txt的时候，发现 https://github.com/huggingface/lerobot@6674e368249472c91382eb54bb8501c94c7f0c56#egg=lerobot这个网址现在已经打不开了，请问您那里有什么解决方法吗。

WPS852, 2025-11-10:
同问，请问解决了吗

rakhimovv, 2026-08-31:
The commit itself is still reachable — `6674e368249472c91382eb54bb8501c94c7f0c56` resolves fine (committed 2025-02-01), so the URL isn't the problem. The install fails one step later, on that commit's dependency metadata:

```
ERROR: Could not find a version that satisfies the requirement pyav>=12.0.5 (from lerobot) (from versions: none)
ERROR: No matching distribution found for pyav>=12.0.5
```

`lerobot` at that pin requires `pyav>=12.0.5`, but the `pyav` project on PyPI publishes **no distributions at all** — `https://pypi.org/simple/pyav/` returns an empty link list (the real package is named `av`). So this requirement cannot be satisfied from PyPI or from any mirror of it, which is why it fails consistently rather than intermittently.

Workaround: drop the `lerobot` line from `requirements.txt`. Nothing in the `VLABench` package imports it — the only consumer is `scripts/convert_to_lerobot.py`, a dataset-conversion utility that isn't part of evaluation. With that line dropped the rest of `requirements.txt` installs cleanly, and evaluation runs normally: I ran all five tracks on Linux/CUDA with EGL rendering. (I installed `rrt-algorithms` separately via the `third_party` step from the README rather than from its requirements line.)

One more thing that bites immediately after this, in case it saves someone a round trip: the README installs `rrt-algorithms` with `--no-deps`, so its `rtree` dependency is never pulled in and the first `from VLABench.tasks import *` fails with

```
File ".../rrt_algorithms/rrt/tree.py", line 1, in <module>
    from rtree import index
ModuleNotFoundError: No module named 'rtree'
```

`pip install rtree` fixes that one.

xiaokang-99, 2026-09-01:
Thanks a lot for your helpful reply! Your suggestion solved my problem. I really appreciate your time and support. 😊